faebryk/faebryk #200: Kicad: Full kicad8 PCB support

IoannisP-ITENG, 2024-06-29:
### Feature Request

We fixed some pcb stuff for kicad8. Not sure how many more problems there are.
Keeping this here till we have more tests.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

ruben-iteng, 2024-06-30:
gr_text is different:

old
```sexpression
(gr_text "TEST123"
    (42.8 26.99)
    (layer "F.SilkS")
    (effects
        (font
            (size 5 5)
            (thickness 0.2)))
    (tstamp "44357"))
```
new
```sexpression
(gr_text "TEST123"
    (at 42.8 26.99 0)
    (layer "F.SilkS")
    (uuid "687a9e9f-fdb6-488c-82ea-fe43bba04ac1")
    (effects
        (font
            (size 5 5)
            (thickness 0.2)
        )
    )
)
```

IoannisP-ITENG, 2024-07-02:
Fixed in #201

IoannisP-ITENG, 2024-07-11:
Def fixed in #233

closed 2024-07-11 by IoannisP-ITENG